ssvb/xf86-video-fbturbo #46: Cannot support Xrandr rotation under Allwinner A33 (AccelMethod = CPU)

Icenowy, 2015-10-06:
xrandr --output default --rotate inverted returns "xrandr: output default cannot use rotation "inverted" reflection "none""

How can I fix it?

Icenowy, 2015-10-06:
I cannot use Rotate=UD now, because it will disable XRANDR, but my window manager (Mutter) do not support X Server without XRANDR.

ssvb, 2015-10-06:
Changing the display output configuration at runtime via XRANDR is not supported at the moment. Not that it can't be done, but this:
* requires some work to implement, which is a dead end because we will move to the KMS driver eventually
* relies on vendor-specific ioctl interface (sunxi disp) even more

I don't know if the A33 kernel from Allwinner has the disp ioctl API, which is fully backwards compatible with A10/A13/A20. For example, do you get Xv overlay working in mplayer? Can you paste your current /var/log/Xorg.0.log somewhere (via gist)?

Icenowy, 2015-10-07:
At least it do not have g2d, so AccelMethod is set to CPU.

And as I know, sunxi disp interface do not support rotation(otherwise I won't refer to X.Org Driver to find a way to rotate the display

---- Siarhei Siamashka编写 ----

Changing the display output configuration at runtime via XRANDR is not supported at the moment. Not that it can't be done, but this:
* requires some work to implement, which is a dead end because we will move to the KMS driver eventually
* relies on vendor-specific ioctl interface (sunxi disp) even more

I don't know if the A33 kernel from Allwinner has the disp ioctl API, which is fully backwards compatible with A10/A13/A20. For example, do you get Xv overlay working in mplayer? Can you paste your current /var/log/Xorg.0.log somewhere (via gist)?

---
https://github.com/ssvb/xf86-video-fbturbo/issues/46#issuecomment-146012185