ssvb/xf86-video-fbturbo #47: eglSwapBuffers

magicse, 2015-10-26:
Driversr 4p0-00rel1
Device Orange Pi2 (Allwinner H3) Mali-400mp2
lsmod 
Module                  Size  Used by
mali_drm                6634  0
drm                   180399  1 mali_drm
mali                  344514  0
ump                    38853  1 mali

In test app triangle I get first frame and after seg fault.
After debug I found that it crash in call eglSwapBuffers();

ssvb, 2015-10-26:
Allwinner H3 is a rather new SoC and you are probably the first one who tried to test 3D acceleration on it in Linux. The r4p0 mali kernel driver also was never used with xf86-video-fbturbo ddx before.

I would suggest to try the xf86-video-mali driver to see if it does not crash for you and report the results.

I also have ordered an Orange Pi PC board with Allwinner H3 myself, but have no idea how long it will take to be delivered from China. Hopefully before Christmas.

magicse, 2015-10-26:
Ok. You driver work normal and not crashed when starting X.
Problems that I found in es2_info seg fault on call XCloseDisplay(x_dpy);
And in triangle test seg fault after first frame on call  eglSwapBuffers(), but I see first frame that mean is problem with buffers.

magicse, 2015-10-26:
xf86-video-mali driver crashed on start X with next log in Xorg 
after satrt a got next errors in Xorg.0.log

 

 

 

[
	13.931] (II) Loading sub module "exa"
[	13.932] (II) LoadModule: "exa"
[	13.932] (II) Loading /usr/lib/xorg/modules/libexa.so
[	13.937] (II) Module exa: vendor="X.Org Foundation"
[	13.937]	compiled for 1.15.1, module version = 2.6.0
[	13.937]	ABI class: X.Org Video Driver, version 15.0
[	13.937] (II) MALI(0): [maliSetupExa:547]: Entering
[	13.938] (II) MALI(0): Mali EXA driver is loaded successfully
[	13.938] (II) MALI(0): [maliSetupExa:591]: Exiting
[	13.938] (WW) MALI(0): [MaliScreenInit:1355] Warning: Initializing EXA Driver!
[	13.938] (II) EXA(0): Driver allocated offscreen pixmaps
[	13.938] (II) EXA(0): Driver registered support for the following operations:
[	13.938] (II)     	Solid
[	13.938] (II)     	Copy
[	13.938] (II)     	Composite (RENDER acceleration)
[	13.938] (==) MALI(0): Backing store enabled
[	13.938] (==) MALI(0): Silken mouse enabled
[	13.945] (II) MALI(0): Changing mode to 720 576 720 1152
[	13.945] (II) MALI(0): RandR 1.2 enabled, ignore the following RandR disabled message.
[	13.946] (--) RandR disabled
[	13.978] (II) SELinux: Disabled on system
[	13.984] (EE) AIGLX error: dlopen of /usr/lib/arm-linux-gnueabihf/dri/Mali DRI2_dri.so failed (/usr/lib/arm-linux-gnueabih

f/dri/Mali DRI2_dri.so: cannot open shared object file: No such file or directory)
[	13.984] (EE) AIGLX: reverting to software rendering
[	14.095] (II) AIGLX: Loaded and initialized swrast
[	14.095] (II) GLX: Initialized DRISWRAST GL provider for screen 0
[	14.097] (EE) MALI(0): [maliModifyPixmapHeader:184] Error: GET_UMP_SECURE_ID_BUF1 returned 0xffffffff offset: 0 virt addre

ss: 0xb63c6000 fb_virt: 0xb63c6000

 
[	14.097] (EE) MALI(0): [maliModifyPixmapHeader:196] Error: UMP failed to retrieve secure id
[	14.097] (II) MALI(0): Setting screen physical size to 190 x 152
[	14.097] (II) MALI(0): fbdev_crtc_config_resize: width = 720 height = 576

 
[	14.097] (EE)
[	14.097] (EE) Backtrace:
[	14.097] (EE)
[	14.097] (EE) Segmentation fault at address 0x20
[	14.097] (EE)

Fatal server error:
[	14.097] (EE) Caught signal 11 (Segmentation fault). Server aborting
[	14.098] (EE)
[	14.098] (EE)

ssvb, 2015-10-26:
Yes, The XCloseDisplay is an old mali gles blob problem, which they never fixed. It indeed may cause segfaults on exit from applications.

Anyway, if the crashes are on the client side, then there is probably nothing to fix in xf86-video-fbturbo which runs in the X server process.

ssvb, 2015-10-26:
> xf86-video-mali driver crashed on start X with next log in Xorg
> after satrt a got next errors in Xorg.0.log

Thanks for this information. It looks like your kernel is probably not implementing the GET_UMP_SECURE_ID_BUF1 ioctl, this upsets xf86-video-mali and makes it crash in a spectacular way. The xf86-video-fbturbo ddx is handling this more gracefully.

magicse, 2015-10-26:
If need I could make more logs and debugs.
Now I try resolve eglSwapBuffers issue.

magicse, 2015-10-30:
Back traces from running test app colour triangle.
(gdb) bt
#0  __GI___libc_malloc (bytes=<optimized out>) at malloc.c:2907
#1  0xb6f59cc4 in __egl_mali_post_to_window_surface () from /usr/lib/libEGL.so
#2  0xb6f59294 in _egl_swap_buffers () from /usr/lib/libEGL.so
#3  0xb6f5836c in eglSwapBuffers () from /usr/lib/libEGL.so
#4  0x00009050 in Redraw (width=200, height=200) at test.c:135
#5  0x00009790 in main (argc=1, argv=0xbefff6f4) at test.c:328

Single stepping until exit from function _egl_surface_job_incref,
which has no line number information.
0xb6f59dbe in __egl_mali_post_to_window_surface () from /usr/lib/libEGL.so
(gdb) s
Single stepping until exit from function __egl_mali_post_to_window_surface,
which has no line number information.

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0xb69d1460 (LWP 2377)]
0xb6f5a224 in _egl_surface_job_decref () from /usr/lib/libEGL.so
(gdb) bt
#0  0xb6f5a224 in _egl_surface_job_decref () from /usr/lib/libEGL.so
#1  0xb6f596be in __egl_worker_callback () from /usr/lib/libEGL.so
#2  0xb6f62238 in _mali_base_worker_thread () from /usr/lib/libEGL.so
#3  0xb6d10fbc in start_thread (arg=0xb69d1460) at pthread_create.c:314
#4  0xb6dc720c in ?? ()
    at ../ports/sysdeps/unix/sysv/linux/arm/nptl/../clone.S:92
   from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)

magicse, 2015-11-11:
With patched blob drivers worked well
http://orangepi.pp.ua/index.php/topic,17.msg569.html#msg569