ssvb/xf86-video-fbturbo #48: G2D/Mixer

magicse, 2015-10-28:
I cant found anything about G2D/Mixer on Allwinner H3
G2D driver used SUNXI_MP_PBASE and SUNXI_IRQ_MP, 
but irqs-sun8iw7p1.h and platform-sun8iw7p1.h didn't have anything like that.

ssvb, 2015-10-29:
It's very likely that H3 does not have the G2D/Mixer hardware block anymore. For example, A13 did not have it either.

magicse, 2015-10-29:
But in datasheet on H3 I found something about RT-MIXER0 RT-MIXER1 in display engine.

ssvb, 2015-10-31:
The references to "mixer" look like something that might be probably related to audio handling in H3.

You can check the A10 manual to find the documentation about the G2D/Mixer that is used for 2D graphics.

magicse, 2015-11-01:
[image]

ssvb, 2015-11-01:
OK, if the picture is correct, then maybe it does have G2D. Do the kernel sources from the Allwinner's H3 SDK have anything interesting?

magicse, 2015-11-01:
display engine 2.0 realtime mixer processing base functions implement
de_rtmx.h
de_rtmx_type.h
de_rtmx.c

ssvb, 2015-11-10:
OK, got an Orange Pi PC board myself and also took a look at the SDK sources. These RT-Mixer and MUX things seem to be used instead of the DEBE and DEFE in older Allwinner chips. As for G2D, it is enabled with the following kernel option:
<pre>
config SUNXI_G2D
	depends on (ARCH_SUN8IW1P1 || ARCH_SUN9IW1P1)
	tristate "sunxi g2d driver"
	default N
	select ION
	select ION_SUNXI
	help
</pre>
Which means A31 or A80, but not H3.