ssvb/xf86-video-fbturbo #51: AIGLX error with RK3066

xoox, 2016-07-30:
I just installed this driver (both master and mali-r3p2-support branch), but it sames failed. The parts of xorg log showing as here.

```
[ 79501.923] (II) FBTURBO(0): using backing store heuristics
[ 79501.928] (II) FBTURBO(0): can't load 'g2d_23' kernel module
[ 79501.928] (II) FBTURBO(0): failed to enable the use of sunxi display controller
[ 79501.928] (II) FBTURBO(0): No sunxi-g2d hardware detected (check /dev/disp and /dev/g2d)
[ 79501.928] (II) FBTURBO(0): G2D hardware acceleration can't be enabled
[ 79501.928] (II) FBTURBO(0): enabled VFP/NEON optimizations
[ 79501.928] (==) FBTURBO(0): Backing store enabled
[ 79501.928] (==) FBTURBO(0): DPMS enabled
[ 79501.928] (II) FBTURBO(0): failed to enable hardware cursor
[ 79501.942] (II) FBTURBO(0): can't load 'sunxi_cedar_mod' kernel module
[ 79501.942] (II) Loading sub module "dri2"
[ 79501.942] (II) LoadModule: "dri2"
[ 79501.942] (II) Module "dri2" already built-in
[ 79501.942] (II) FBTURBO(0): display controller hardware overlays can't be used for DRI2
[ 79501.943] (II) FBTURBO(0): Wait on SwapBuffers? enabled
[ 79501.943] (II) FBTURBO(0): [DRI2] Setup complete
[ 79501.943] (II) FBTURBO(0): [DRI2]   DRI driver: lima
[ 79501.943] (II) FBTURBO(0): using DRI2 integration for Mali GPU (UMP buffers)
[ 79501.943] (II) FBTURBO(0): Mali binary drivers can only accelerate EGL/GLES
[ 79501.943] (II) FBTURBO(0): so AIGLX/GLX is expected to fail or fallback to software
[ 79501.943] (==) RandR enabled
[ 79501.962] (II) SELinux: Disabled on system
[ 79501.965] (EE) AIGLX error: dlopen of /usr/lib/arm-linux-gnueabihf/dri/lima_dri.so failed (/usr/lib/arm-linux-gnueabihf/dri/lima_dri.so: cannot open shared object file: No such file or directory)
[ 79501.965] (EE) AIGLX: reverting to software rendering
```

Would you please give any idea to fix this?

Here's some extra information.
Board: [Marsboard RK3066](http://marsboard.org/marsboard_rk3066_feature.html)
OS: Debian 8.5
kernel: [rockchip-marsboard-rk3066-linux-3.0.8](https://github.com/powermcu/rockchip-marsboard-rk3066-linux-3.0.8)
mali kernel driver: integrated to the kernel. version r3p2.
[sunxi-mali](https://github.com/linux-sunxi/sunxi-mali) installed with VERSION=r3p2-01rel1

ssvb, 2016-07-30:
This is not a problem. Mali drivers don't provide GLX support, and offer EGL instead. This "error" message in the log is just a noise coming from the X server because the xf86-video-fbturbo can't suppress it nicely and the X server has some desktopish OpenGL assumptions. The current workaround is just printing something like ```"ignore the AIGLX error message, it is not an error"``` in the log later on. And you have cut off the relevant part of the log.

ssvb, 2016-07-30:
Have you tried to run ```es2_info```, ```es2gears``` or ```glmark2-es2```? Why do you think that the driver failed?

xoox, 2016-07-31:
Thanks for your explanation.
I'll have a check with *es2* tools.

I got error messages about GLX failure, and take it for granted that problems happened.

closed 2016-07-31 by xoox