ssvb/xf86-video-fbturbo #52: cant compile

hmwel, 2016-08-22:
CCLD     fbturbo_drv.la
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/5/../../../arm-linux-gnueabihf/libUMP.a(ump_arch.o): relocation R_ARM_THM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/usr/lib/gcc/arm-linux-gnueabihf/5/../../../arm-linux-gnueabihf/libUMP.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
Makefile:484: recipe for target 'fbturbo_drv.la' failed
make[2]: *** [fbturbo_drv.la] Error 1

compling on arm using Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.2 

YES i tried compiling using -fPIC

ssvb, 2016-08-22:
How did you install libUMP?

hmwel, 2016-08-22:
i installed it following this guide. 
http://wiki.lemaker.org/BananaPro/Pi:GPU

belowe the output of the install 
`sudo make install
[sudo] password for : 
Making install in src
make[1]: Entering directory '/home/user/hw_accleration/libump/src'
make[2]: Entering directory '/home/user/hw_accleration/libump/src'
 /bin/mkdir -p '/usr/local/lib'
 /bin/bash ../libtool   --mode=install /usr/bin/install -c   libUMP.la '/usr/local/lib'
libtool: install: /usr/bin/install -c .libs/libUMP.so.3.0.0 /usr/local/lib/libUMP.so.3.0.0
libtool: install: (cd /usr/local/lib && { ln -s -f libUMP.so.3.0.0 libUMP.so.3 || { rm -f libUMP.so.3 && ln -s libUMP.so.3.0.0 libUMP.so.3; }; })
libtool: install: (cd /usr/local/lib && { ln -s -f libUMP.so.3.0.0 libUMP.so || { rm -f libUMP.so && ln -s libUMP.so.3.0.0 libUMP.so; }; })
libtool: install: /usr/bin/install -c .libs/libUMP.lai /usr/local/lib/libUMP.la
libtool: install: /usr/bin/install -c .libs/libUMP.a /usr/local/lib/libUMP.a
libtool: install: chmod 644 /usr/local/lib/libUMP.a
libtool: install: ranlib /usr/local/lib/libUMP.a
libtool: finish: PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/sbin" ldconfig -n /usr/local/lib
----------------------------------------------------------------------
Libraries have been installed in:
   /usr/local/lib

If you ever happen to want to link against installed libraries
in a given directory, LIBDIR, you must either use libtool, and
specify the full pathname of the library, or use the '-LLIBDIR'
flag during linking and do at least one of the following:
   - add LIBDIR to the 'LD_LIBRARY_PATH' environment variable
     during execution
   - add LIBDIR to the 'LD_RUN_PATH' environment variable
     during linking
   - use the '-Wl,-rpath -Wl,LIBDIR' linker flag
   - have your system administrator add LIBDIR to '/etc/ld.so.conf'

See any operating system documentation about shared libraries for
more information, such as the ld(1) and ld.so(8) manual pages.
----------------------------------------------------------------------
 /bin/mkdir -p '/usr/local/include/ump/'
 /usr/bin/install -c -m 644 ump.h ump_platform.h ump_ref_drv.h '/usr/local/include/ump/'
make[2]: Leaving directory '/home/user/hw_accleration/libump/src'
make[1]: Leaving directory '/home/user/hw_accleration/libump/src'
make[1]: Entering directory '/home/user/hw_accleration/libump'
make[2]: Entering directory '/home/user/hw_accleration/libump'
make[2]: Nothing to be done for 'install-exec-am'.
 /bin/mkdir -p '/usr/local/lib/pkgconfig'
 /usr/bin/install -c -m 644 libump.pc '/usr/local/lib/pkgconfig'
make[2]: Leaving directory '/home/user/hw_accleration/libump'
make[1]: Leaving directory '/home/user/hw_accleration/libump'
`

ssvb, 2016-08-22:
Was it libUMP that you recompiled with -fPIC?

ssvb, 2016-08-22:
But your libUMP seems to be getting installed into /usr/local/lib, is this correct?

hmwel, 2016-08-22:
sudo ls ```
/usr/local/lib                                                                                                                                                                                                               1 ↵
libdri2.a  libdri2.la  libdri2.so  libdri2.so.1  libdri2.so.1.0.0  libnl.so  libnl.so.1  libnl.so.1.1  libUMP.a  libUMP.la  libUMP.so  libUMP.so.3  libUMP.so.3.0.0  pkgconfig  python2.7  python3.5

```
I guess its installed there i did not compiled libump with -fPIC ( atm trying)

ssvb, 2016-08-22:
But GCC seems to be complaining about  ``/usr/lib/gcc/arm-linux-gnueabihf/5/../../../arm-linux-gnueabihf/libUMP.a``, which does not reside under ``/usr/local``. Could it be that your system already has some other (broken?) libUMP installed in ``/usr/lib``?

hmwel, 2016-08-22:
that can be but if it is taking it from /usr/lib/gcc/arm-linux-gnueabihf/5/../../../arm-linux-gnueabihf/libUMP.a it is using  the ubuntu gcc package

hmwel, 2016-08-22:
Compiling libump with -fPIC fixed the problem

Tnx for your help