ssvb/xf86-video-fbturbo #54: cant do make

Laksa19, 2017-04-25:
root@orangepipc:/home/laksa/xf86-video-fbturbo# ./configure --prefix=/usr 
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking for style of include used by make... GNU
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for gcc option to accept ISO C99... none needed
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking whether __clang__ is declared... no
checking whether __INTEL_COMPILER is declared... no
checking whether __SUNPRO_C is declared... no
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking build system type... armv7l-unknown-linux-gnueabihf
checking host system type... armv7l-unknown-linux-gnueabihf
checking for a sed that does not truncate output... /bin/sed
checking if gcc supports -Werror=unknown-warning-option... no
checking if gcc supports -Werror=unused-command-line-argument... no
checking if gcc supports -Wall... yes
checking if gcc supports -Wpointer-arith... yes
checking if gcc supports -Wmissing-declarations... yes
checking if gcc supports -Wformat=2... yes
checking if gcc supports -Wstrict-prototypes... yes
checking if gcc supports -Wmissing-prototypes... yes
checking if gcc supports -Wnested-externs... yes
checking if gcc supports -Wbad-function-cast... yes
checking if gcc supports -Wold-style-definition... yes
checking if gcc supports -Wdeclaration-after-statement... yes
checking if gcc supports -Wunused... yes
checking if gcc supports -Wuninitialized... yes
checking if gcc supports -Wshadow... yes
checking if gcc supports -Wmissing-noreturn... yes
checking if gcc supports -Wmissing-format-attribute... yes
checking if gcc supports -Wredundant-decls... yes
checking if gcc supports -Wlogical-op... yes
checking if gcc supports -Werror=implicit... yes
checking if gcc supports -Werror=nonnull... yes
checking if gcc supports -Werror=init-self... yes
checking if gcc supports -Werror=main... yes
checking if gcc supports -Werror=missing-braces... yes
checking if gcc supports -Werror=sequence-point... yes
checking if gcc supports -Werror=return-type... yes
checking if gcc supports -Werror=trigraphs... yes
checking if gcc supports -Werror=array-bounds... yes
checking if gcc supports -Werror=write-strings... yes
checking if gcc supports -Werror=address... yes
checking if gcc supports -Werror=int-to-pointer-cast... yes
checking if gcc supports -Werror=pointer-to-int-cast... yes
checking if gcc supports -pedantic... yes
checking if gcc supports -Werror... yes
checking if gcc supports -Werror=attributes... yes
checking whether make supports nested variables... (cached) yes
checking dependency style of gcc... gcc3
checking how to print strings... printf
checking for a sed that does not truncate output... (cached) /bin/sed
checking for fgrep... /bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert armv7l-unknown-linux-gnueabihf file names to armv7l-unknown-linux-gnueabihf format... func_convert_file_noop
checking how to convert armv7l-unknown-linux-gnueabihf file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /bin/dd
checking how to truncate binary pipes... /bin/dd bs=4096 count=1
checking for mt... mt
checking if mt is a manifest tool... no
checking for dlfcn.h... yes
checking for objdir... .libs
checking if gcc supports -fno-rtti -fno-exceptions... no
checking for gcc option to produce PIC... -fPIC -DPIC
checking if gcc PIC flag -fPIC -DPIC works... yes
checking if gcc static flag -static works... yes
checking if gcc supports -c -o file.o... yes
checking if gcc supports -c -o file.o... (cached) yes
checking whether the gcc linker (/usr/bin/ld) supports shared libraries... yes
checking whether -lc should be explicitly linked in... no
checking dynamic linker characteristics... GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... no
./configure: line 18655: syntax error near unexpected token `RANDR,'
./configure: line 18655: `XORG_DRIVER_CHECK_EXT(RANDR, randrproto)'
root@orangepipc:/home/laksa/xf86-video-fbturbo# make
make: *** No targets specified and no makefile found.  Stop.
root@orangepipc:/home/laksa/xf86-video-fbturbo#

ssvb, 2017-04-25:
How did you generate the ``configure`` script?

whirmill, 2017-06-25:
I was having the same problem on archlinux, solved installing xorg-server-devel

Olm-e, 2019-03-27:
I had the same problem with an A20 board on debian jessie

apt install xorg-dev 
autoreconf -vi 
./configure 
make
sudo make install 

does seems to solve it ;)