ssvb/xf86-video-fbturbo #57: syntax error near unexpected token `RENDER,'

ab1jx, 2018-05-24:
I'm on a Raspberry Pi 3B running Debian Buster arm64, a little unusual at this point.  I run autogen.sh and I get:

./configure: line 18657: syntax error near unexpected token `RENDER,'
./configure: line 18657: `XORG_DRIVER_CHECK_EXT(RENDER, renderproto)'

I got it on the RANDR line so I commented that out as a test.  As far as I can tell I have reasonably complete randr stuff with the dev packages.  The error just moved to the next line.  Should the driver checks cause syntax errors?

I use fbturbo under Raspbian, the Debian comes with just fbdev.  I may be seeing issues with fbdev so I was looking for an alternative, but if this if forked from it it may not help anyway.

Attaching the configure that autogen.sh made (renamed to suit the attaching/uploading wizard that looks at file extensions.  And config.log
[config.log](https://github.com/ssvb/xf86-video-fbturbo/files/2035544/config.log)

[configure.txt](https://github.com/ssvb/xf86-video-fbturbo/files/2035538/configure.txt)
)

ssvb, 2018-05-24:
Have you installed all the build dependencies as explained at https://github.com/ssvb/xf86-video-fbturbo/wiki/Installation ?

ab1jx, 2018-05-24:
Um, nope, I'd only just found fbturbo, didn't realize there was a wiki.  I grabbed master.zip and tried to build it.  I'd only read the README.  That did it, built and installed.

closed 2018-05-24 by ab1jx